microsoft/AutonomousDrivingCookbook #130: AD_Cookbook_AirSim.7z download link do not work! help!

xiaoToby, 2023-11-16:
**Your issue may already be reported! Please make sure to search all open and closed issues before starting a new one.**

Please fill out the sections below so we can understand your issue better and resolve it quickly.

## Problem description
*(Please provide a 2-3 sentence description of your problem. Be concise to ensure this description is useful for future users who might run into the same issue.)*

## Problem details
*(Please describe your problem in as much detail as possible here. Make sure to include screenshots, code snippets, error messages, links and anything else you think will help us understand your problem better. If applicable, please also provide us a list of steps to reproduce your problem.)*

## Experiment/Environment details
* Tutorial used: *(For example, AirSimE2EDeepLearning, DistributedRL etc.)*
* Environment used: *(For example, landscape, city, hawaii etc.)*
* Versions of artifacts used (if applicable): *(For example, Python 3.5, Keras 2.1.2 etc.)*

xiaoToby, 2023-11-20:
@mitchellspryn hello, i know you are not work for this repo, but there is nobody that I can find to ask a help for AD_Cookbook_AirSim.7z , the link for that binaries files did not work, so I can not download that to go on my research for that repo, if you still have that file, could you send it to me ?

Praveenx1, 2024-02-09:
I am facing the same issue!!
Let me know if it was resolved

xiaoToby, 2024-02-19:
you should find someone who has these files to send you